adafruit/Adafruit_TLC59711 #8: Library working fine for arduino but not for esp32

harshcht, 2017-06-14:
The library is working fine for my arduino mega board but my esp32 hangs .
it displays the following error : 

Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d1538  PS      : 0x00060530  A0      : 0x800d921e  A1      : 0x3ffc7440  
A2      : 0x3ffc1cb8  A3      : 0x3ffc1e10  A4      : 0x00060023  A5      : 0x3ffc74f8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0663  A9      : 0x3ffc7420  
A10     : 0x3ffc1cb8  A11     : 0x00000000  A12     : 0x0800001c  A13     : 0xffffffff  
A14     : 0xffffffff  A15     : 0x00060023  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1538:0x3ffc7440 0x400d921e:0x3ffc7460

CPU halted.

Can anyone help me out please.

DvDriel, 2019-05-14:
Example code not working for me either on a Adafruit feater esp32, different error code though:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
TLC59711 test
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0e47  PS      : 0x00060130  A0      : 0x800d0ccc  A1      : 0x3ffb1f30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffffe8  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x80081126  A9      : 0x3ffb1f10  
A10     : 0x000087ac  A11     : 0x00000001  A12     : 0x80087b20  A13     : 0x3ffb1e40  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d0e47:0x3ffb1f30 0x400d0cc9:0x3ffb1f50 0x400d0b65:0x3ffb1f80 0x400d212f:0x3ffb1fb0 0x40087911:0x3ffb1fd0

Did you have any luck finding the source of the problem yet?

p1234321, 2020-11-05:
Same Problem here:
`Rebooting...
⸮!⸮⸮ަ⸮⸮⸮⸮ݜ⸮⸮⸮!⸮⸮⸮!ǡ⸮TLC5974 test
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d15a1  PS      : 0x00060f30  A0      : 0x80088218  A1      : 0x3ffb1fb0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb8058  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d159e  A9      : 0x3ffb1f80  
A10     : 0x3ffc02e4  A11     : 0x3ffb0060  A12     : 0x00000020  A13     : 0x80000020  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d15a1:0x3ffb1fb0 0x40088215:0x3ffb1fd0

`
Anybody konows what to do?

ladyada, 2020-11-05:
if folks can add print's to figure out what function is causing it, that will help debug!

p1234321, 2020-11-05:
Thanks for the fast response.
I just used the TLC59711 test sketch.
Everything works fine until it tries to write to tlc the first time:
`#include "Adafruit_TLC59711.h"
#include <SPI.h>
#define NUM_TLC59711 2
#define data   11
#define clock  13

Adafruit_TLC59711 tlc = Adafruit_TLC59711(NUM_TLC59711, clock, data);
void setup() {
  Serial.begin(9600); 
  Serial.println("TLC59711 test");
  tlc.begin();
  //tlc.write();
}
void loop() { 
}`

so when i comment the "tlc.write();" it runs without comlaints, but as soon as it starts the write-function the error occurs.

ladyada, 2020-11-05:
ok not sure wher eit is, maybe interrupts cant be turned off.

p1234321, 2020-11-05:
Did you try with esp32? Does it maybe have to do with memory allocation?

mharriman101, 2020-11-05:
Same issue on Arduino Nano Every - fails of first call to tlc.write in setup function - causes board reset.

p1234321, 2020-11-26:
I think this line is cusing the problem:
_spi->endTransaction();
maybe it helps...

p1234321, 2020-11-26:
Also its not crashing when using " Adafruit_TLC59711 tlc = Adafruit_TLC59711(NUM_TLC59711);" instead of "Adafruit_TLC59711 tlc = Adafruit_TLC59711(NUM_TLC59711, clock, data);"
(so without defining clock and data pins)

p1234321, 2020-11-27:
I think i found the issue:
At this point there where brackets missing:
`  else
    delayMicroseconds(2);
  _spi->endTransaction(); //this should be within the else statement

  interrupts();
}`

it should be like this: 
`else{

    delayMicroseconds(2);
    _spi->endTransaction();   //seems to trigger
  }`

ladyada, 2020-11-28:
hihi please submit a PR :)

ladyada, 2021-06-05:
fixed in #19 ! :)

closed 2021-06-05 by ladyada